mutovis/control-software #2: there is no GUI

greyltc, 2018-10-30:
Controlling the collection of solar cell data is done through interaction with a command line program with the following interface:
```
usage: cli.py [-h] [--destination DESTINATION] [--pixel_address PIXEL_ADDRESS]
              [--sweep SWEEP] [--snaith SNAITH] [--t-prebias T_PREBIAS]
              [--area AREA] [--mppt MPPT] [--t-dwell T_DWELL]
              [--relay-ip RELAY_IP] [--wavelabs WAVELABS] [--rear REAR]
              [--four-wire FOUR_WIRE]
              [--current-compliance-override CURRENT_COMPLIANCE_OVERRIDE]
              [--scan-low-override SCAN_LOW_OVERRIDE]
              [--scan-high-override SCAN_HIGH_OVERRIDE]
              [--scan-points SCAN_POINTS] [--scan-NPLC SCAN_NPLC]
              [--terminator TERMINATOR] [--baud BAUD] [--port PORT]
              [--address ADDRESS] [--switch_address SWITCH_ADDRESS]
              [--visa-lib VISA_LIB] [--dummy] [--scan] [--test-hardware]
              operator

Automated solar cell IV curve collector using a Keithley 24XX sourcemeter.
Data is written to HDF5 files and human readable messages are written to
stderr.

positional arguments:
  operator              Name of operator

optional arguments:
  -h, --help            show this help message and exit
  --destination DESTINATION
                        Save output files here. '__tmp__' will use a system
                        default temporary directory

optional arguments for measurement configuration:
  --pixel_address PIXEL_ADDRESS
                        Hex value to specify an enabled pixel bitmask (must
                        start with 0x...)
  --sweep SWEEP         Do an I-V sweep from Voc --> Jsc
  --snaith SNAITH       Do an I-V sweep from Jsc --> Voc
  --t-prebias T_PREBIAS
                        Number of seconds to sit at initial voltage value
                        before doing sweep
  --area AREA           Specify device area in cm^2
  --mppt MPPT           Do maximum power point tracking for this many cycles
  --t-dwell T_DWELL     Total number of seconds for the dwell mppt phase(s)

optional arguments for setup configuration:
  --relay-ip RELAY_IP   IP address of the WaveLabs relay server (set to 0 for
                        direct WaveLabs connection)
  --wavelabs WAVELABS   WaveLabs LED solar sim is present
  --rear REAR           Use the rear terminals
  --four-wire FOUR_WIRE
                        Use four wire mode (the defalt)
  --current-compliance-override CURRENT_COMPLIANCE_OVERRIDE
                        Override current compliance value used diring I-V
                        scans
  --scan-low-override SCAN_LOW_OVERRIDE
                        Override more negative scan voltage value
  --scan-high-override SCAN_HIGH_OVERRIDE
                        Override more positive scan voltage value
  --scan-points SCAN_POINTS
                        Number of measurement points in I-V curve
  --scan-NPLC SCAN_NPLC
                        Sourcemeter NPLC setting to use during I-V scan
  --terminator TERMINATOR
                        Instrument comms read & write terminator (enter in
                        hex)
  --baud BAUD           Instrument serial comms baud rate
  --port PORT           Port to connect to switch hardware
  --address ADDRESS     VISA resource name for sourcemeter
  --switch_address SWITCH_ADDRESS
                        IP address for PCB
  --visa-lib VISA_LIB   Path to visa library in case pyvisa can't find it, try
                        C:\Windows\system32\visa64.dll

optional arguments for debugging/testing:
  --dummy               Run in dummy mode (doesn't need sourcemeter, generates
                        simulated device data)
  --scan                Scan for obvious VISA resource names, print them and
                        exit
  --test-hardware       Exercises all the hardware, used to check for and
                        debug issues
```
a GUI would be nice!

greyltc, 2019-02-25:
work still in progress
this will be the entry point: https://github.com/mutovis/control-software/blob/master/mutovis-control-gui

greyltc, 2019-02-27:
@rbrenesh
To slightly alleviate this problem today, I've solved https://github.com/mutovis/analysis-software/issues/3
This means that by picking your data collection folder by using the File --> Watch menu item in `mutovis-analysis`, your measurement results will show up in the table automatically as soon as the data collection file has been closed by the measurement program. It's not quite real-time data monitoring, but it's getting at least closer!

rbrenesh, 2019-02-28:
I tried using the Watch mode after updating the program using:

`yay -S python-mutovis-analysis`

I don't know if you have pushed the update to the package manager but when I enable the Watch feature the app is crashing and I'm getting this error:

`Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/batch_iv_analysis/gui.py", line 1182, in handleWatchAction
    self.handleWatchUpdate(self.workingDirectory)
  File "/usr/lib/python3.7/site-packages/batch_iv_analysis/gui.py", line 1233, in handleWatchUpdate
    self.newFiles(new_files_fp)
  File "/usr/lib/python3.7/site-packages/batch_iv_analysis/gui.py", line 899, in newFiles
    self.analyzer.processFiles(fullPaths, self.processFitResult, self.primeRow)
  File "/usr/lib/python3.7/site-packages/batch_iv_analysis/ivAnalyzer.py", line 305, in processFiles
    fileDatas = ivAnalyzer._loadFile(fullPath)
  File "/usr/lib/python3.7/site-packages/batch_iv_analysis/ivAnalyzer.py", line 374, in _loadFile
    iv_data = pixel['all_measurements']          
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "/usr/lib/python3.7/site-packages/h5py/_hl/group.py", line 262, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5o.pyx", line 190, in h5py.h5o.open
KeyError: "Unable to open object (object 'all_measurements' doesn't exist)"
Aborted (core dumped)`

greyltc, 2019-02-28:
hm. looks like maybe there is a malformed file in the directory you're watching (from a run that was aborted before the h5 file was completely written maybe?).

Does watch work if you start watching an empty directory to start?

rbrenesh, 2019-02-28:
Yeah you're totally right. I was doing some test runs and aborted one, once I deleted that run it was back to normal.

greyltc, 2019-03-01:
But the tool needs to be robust to that. Do you still have the bad file you can send me?

greyltc, 2019-03-01:
the crash from https://github.com/mutovis/control-software/issues/2#issuecomment-468447573 is solved: https://github.com/mutovis/analysis-software/issues/7