mutovis/control-software #3: Cannot run layout index 8 and 10

rbrenesh, 2019-02-25:
Running the command for layout index 8 or 10 (example below):
`mutovis-control-cli --layout-index 8 --destination /home/onelab/data --pixel-address A4A5  -o grey -r "buffalo thickness study" -p thickness 1.2m -p "hair color" turquoise`

Returns the error:
```
Traceback (most recent call last):
  File "/usr/bin/mutovis-control-cli", line 11, in <module>
    load_entry_point('mutovis-control==1.0.2', 'console_scripts', 'mutovis-control-cli')()
  File "/usr/lib/python3.7/site-packages/mutovis_control/launch_cli.py", line 6, in main
    cli.run()
  File "/usr/lib/python3.7/site-packages/mutovis_control/cli.py", line 182, in run
    pixel_que = self.buildQ(args.pixel_address)
  File "/usr/lib/python3.7/site-packages/mutovis_control/cli.py", line 438, in buildQ
    area = using_layouts[this_substrate]['pixelareas'][this_pixel]
TypeError: 'NoneType' object is not subscriptable
```

greyltc, 2019-02-26:
I can recreate this.

greyltc, 2019-02-26:
The problem was with any layout with less than 6 pixels. It should be fixed by 48495ecc3a8ed7229684e4d84759b7109c0d28b4

closed 2019-02-26 by greyltc

greyltc, 2019-02-26:
@rbrenesh, I've made a new release that contains a fix for this as well as the one in #5. You can grab and install it with 
```
yay -S python-mutovis-control
```