mutovis/control-software #4: the mppt can get stuck in a loop

greyltc, 2019-02-26:
````
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
WARNING: Reached high voltage edge because we hit Voc
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
Reached low voltage edge because angle exceeded
```

greyltc, 2019-02-26:
Looks like probably this only happens with a disconnected pixel. ie. it's not responding as expected during the exploration phases.

rbrenesh, 2019-03-14:
I ran into a similar problem while MPPT a device over 2 hours. However, it seems like the Keithley hit the compliance limit for the current. Attached is a picture of the command line when it happened.

However, the device was still operational after this with a much reduced Vmpp of 0.3V vs ~0.8V where it started. The device wasn't shorted as far as I could tell.

[image]

greyltc, 2019-03-21:
First off, I'm surprised the Keithley hit the compliance limit during the MPPT. I set the compliance here to be 2x the value for the current at the maximum power point as found during a previous J-V sweep. I think probably the MPPT algorithm erroneously went beyond Voc and fully turned on the device's diode which is why the current went into compliance.

I've made changes that I hope will prevent endless loops while MPPTing. In 06853af7e8cfc9aff5f2693a627859620435b6a7 I've put some checks in that should prevent the basic MPPT algorithm from getting stuck behind "edges" that it should transition off of during the exploration phases.

Let's re-open this if we see any new endless loops.

closed 2019-03-21 by greyltc