mutovis/control-software #7: Unable to download latest release

rbrenesh, 2019-03-06:
I'm unable to download [`28ead5b`](https://github.com/mutovis/control-software/commit/28ead5b44e1e6547432dc29f6ebcb49e92a26ba5) in Arch using the regular command:

`yay -S python-mutovis-control`

When running it, I get the error:

```
==> Making package: python-mutovis-control 1.0.6-1 (Tue 05 Mar 2019 04:07:04 PM EST)
==> Retrieving sources...
  -> Found v1.0.6.tar.gz
==> Validating source files with sha256sums...
    v1.0.6.tar.gz ... FAILED
==> ERROR: One or more files did not pass the validity check!
Error downloading sources: python-mutovis-control
```

greyltc, 2019-03-06:
Oops, I had an internet connectivity glitch while I was calculating the checksum for the package so maybe it got borked. I've re-calculated and updated it. Try now.

And just to be safe, when yay asks you if you want do do a clean build of it, do it.

rbrenesh, 2019-03-06:
Worked!

closed 2019-03-06 by rbrenesh