mutovis/control-software #8: the mppt's exploration region is too large

greyltc, 2019-03-14:
@rbrenesh
Creating this issue to capture previous email

rbrenesh, 2019-03-14:
Adding a picture of 30s MPPT of a device for context. It seems like the lower end of the exploration is a bit too big. Reducing that should also make the exploration phase shorter as well.

[image]

greyltc, 2019-03-15:
I'll make the size of the exploration region something that can be configured. I'm also working on adding a new mppt algorithm

greyltc, 2019-03-20:
@rbrenesh,
03877e084ac815c0870b578ae4acdda4ceb71b60 should address this, it makes the size of the exploration region configurable (as well as the duration of the dwell phases, which were previously fixed at 10s).

It adds a new interface for configuring the max power point tracker with `--mppt-params` This is also how we will switch between different maximum power point tracking algorithms and configure them in the future, right now the only mppt available is `basic`.

The old mppt was equivalent to `--mppt-params basic://7:10` that means +/- 7 degrees of exploration and 10 second dwell periods. Now for example you can do `--mppt-params basic://3:10` which means you'll explore +/- 3 degrees.

Here's +/- 7 degrees (the previous unconfigurable MPPT) for 37 seconds:
[image]
and here's +/- 3 degrees with `--mppt-params basic://3:10` also for 37 seconds:
[image]
Don't mind the sloped power output on these graphs, that's just the battery dying on the flashlight I use to illuminate & test at my desk! ;-)
 
I think this should give you what you were looking for! If not, feel free to just comment here to let me know what I've missed.

I've released this in v1.0.8, so as usual
`yay -Sy python-mutovis-control`

closed 2019-03-20 by greyltc