mutovis/control-software #9: there's no way to eliminate the mppt's exploration phase from the output data

greyltc, 2019-03-14:
@rbrenesh
creating this issue to capture previous email

greyltc, 2019-03-14:
How about I add an extra column to the data file, "exploring" that's 1 when the mppt is in the explore phase and 0 when its in the dwell phase?

Then you could use that column to eliminate the exploration phase data from your plots when you do your data analysis.

rbrenesh, 2019-03-14:
I think that's a good solution.

Is there a way to make measurements without saving the data? I was going to fork the code and try a few different MPPT implementations like the ones outlined [here](http://ww1.microchip.com/downloads/en/appnotes/00001521a.pdf). That way, MPPT is done actively at all time steps with no need for an exploration phase.

greyltc, 2019-03-15:
About making measurements without saving data, at present the measurement data is always saved. I guess I don't understand how not saving it would be helpful.

I'm 100% with you. The current MPPT is total crap. One that operates continuously would be better for sure and that document looks like a great place to start. Absolutely fork away and play with different mppt implementations. I'm working on a new/better mppt algorithm too. I'll include it as soon as I get it going. I'll probably change up the command line interface to make switching between mppt algorithms an option.

greyltc, 2019-03-21:
@rbrenesh,
In thinking about this more, I think the real problem is just that the basic MPPT algorithm is garbage. I'd like to not include the "exploring" column in the output data because that only applies to this one MPPT algorithm and feels pretty wrong to include a special column in the output file that either disappears or does not apply when another MPPT algorithm is in use.

So I've added a new gradient descent MPPT algorithm that operates entirely differently and does not have the problem of wild power swings in special exploration phases. I've made some new documentation on how to switch between and configure the different MPPT algorithms: https://github.com/mutovis/control-software/blob/master/MPPT.md

So I think this issue is solved by using the new algorithm by appending `--mppt-params gradient_descent://0.001` to your measurement command. If this does not solve the issue, let me know and we'll re-open this.

greyltc, 2019-03-21:
gradient_descent works in the latest release, v1.1.0 so
`yay -Sy python-mutovis-control`

closed 2019-03-21 by greyltc

rbrenesh, 2019-03-21:
Awesome! I'm leaving for spring break so I won't be able to test it myself but I'll ask some of the device people to try it out and tell me how it goes. I'll keep you updated!