mutovis/control-software #11: A6 not being measured with layout index 8

rbrenesh, 2019-03-16:
Whenever I specify device A6 to be measured I get the line (see attached picture as well).

`INFO: Skipping zero area pixel: A6`

I checked  the usr/etc/layouts.ini file and it seems to be correct though... (see second picture)

[image]

[image]

greyltc, 2019-03-16:
Oh, that sounds like a bug!

greyltc, 2019-03-20:
Yep, I can recreate this

greyltc, 2019-03-20:
@rbrenesh,
Should be all fixed by 5810841e5f474bfe0800ca181a1e33f4d78bcf42
and packaged up in v1.0.7
`yay -Sy python-mutovis-control`

closed 2019-03-20 by greyltc