mutovis/control-software #12: Update 1.0.7 fails to install

rbrenesh, 2019-03-20:
Running the command ` yay -Sy python-mutovis-control`

returns the error:

`:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                     [######################] 100%
(1/1) checking package integrity                   [######################] 100%
(1/1) loading package files                        [######################] 100%
(1/1) checking for file conflicts                  [######################] 100%
error: failed to commit transaction (conflicting files)
python-pyusb: /usr/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg-info/PKG-INFO exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg-info/SOURCES.txt exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg-info/dependency_links.txt exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/pyusb-1.0.2-py3.7.egg-info/top_level.txt exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__init__.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/__init__.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/_debug.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/_interop.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/_lookup.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/_objfinalizer.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/control.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/core.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/legacy.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/libloader.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/__pycache__/util.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/_debug.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/_interop.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/_lookup.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/_objfinalizer.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/backend/__init__.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/backend/__pycache__/__init__.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/backend/__pycache__/libusb0.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/backend/__pycache__/libusb1.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/backend/__pycache__/openusb.cpython-37.pyc exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/backend/libusb0.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/backend/libusb1.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/backend/openusb.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/control.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/core.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/legacy.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/libloader.py exists in filesystem
python-pyusb: /usr/lib/python3.7/site-packages/usb/util.py exists in filesystem
Errors occurred, no packages were upgraded.
Error installing repo packages`

rbrenesh, 2019-03-20:
Fixed, the package pyftdi was installed with pip before so it was conflicting with this one. I just uninstalled the one installed with pip and it's now working correctly.

closed 2019-03-20 by rbrenesh

greyltc, 2019-03-20:
oh no! please don't ever use pip here! __especially__ don't use pip with `sudo`. pip operates independently of Arch's package manager (pacman) and can cause all kinds of conflicts like this by putting whatever files it decides to, wherever it wants to on your machine.

But luckily if things ever get really messed up, you should always be able to start fresh by reinstalling a new Arch based OS and running
```
sudo pacman -Syu ansible --needed
ansible-pull -U https://github.com/mutovis/deploy --ask-become-pass
```
as described in https://github.com/mutovis/deploy

I try pretty hard to let all the system-wide software to be managed by [pacman](https://wiki.archlinux.org/index.php/pacman) (Arch's package manager). The `yay` command uses pacman under the hood. I've packaged the analyzer`[1]` and control software`[2]` in the AUR (Arch User Repository) so that they (and all their dependencies) get installed, and updated properly (hopefully without conflicts like this) with the rest of the system software using yay (pacman).

`[1]`:  https://aur.archlinux.org/packages/python-mutovis-analysis/
`[2]`: https://aur.archlinux.org/packages/python-mutovis-control/