mutovis/control-software #17: Version 1.1.3 fails to run

rbrenesh, 2019-04-09:
Running the test command:

`mutovis-control-cli --layout-index 8 --destination /home/onelab/data --pcb-address 192.168.1.11:23 --light-address ftdi://ftdi:232/1 --sm-address GPIB0::24::INSTR --rear false --four-wire on --pixel-address A1 --sweep on --snaith on --mppt-10 gradient_descent://10:0.001  -o grey -r "buffalo thickness study" -p thickness 1.2m -p "hair color" turquoise`

Returns the error:

`Traceback (most recent call last):
  File "/usr/bin/mutovis-control-cli", line 11, in <module>
    load_entry_point('mutovis-control==1.1.3', 'console_scripts', 'mutovis-control-cli')()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3.7/site-packages/mutovis_control/__init__.py", line 7, in <module>
    from .fabric import fabric
  File "/usr/lib/python3.7/site-packages/mutovis_control/fabric.py", line 1, in <module>
    import h5py
  File "/usr/lib/python3.7/site-packages/h5py/__init__.py", line 36, in <module>
    from ._conv import register_converters as _register_converters
  File "h5py/h5r.pxd", line 21, in init h5py._conv
  File "h5py/_objects.pxd", line 12, in init h5py.h5r
  File "h5py/_objects.pyx", line 1, in init h5py._objects
ImportError: /usr/lib/python3.7/site-packages/h5py/defs.cpython-37m-x86_64-linux-gnu.so: undefined symbol: H5Ovisit_by_name`

Seems to be some problem with h5py.... Looking into it

rbrenesh, 2019-04-09:
Uninstalled hdf5-java and hdf5-view and installed hdf5. Seems to have resolved the issue

closed 2019-04-09 by rbrenesh

greyltc, 2019-04-11:
That's unexpected. I run with the `hdf5-java` package on all my machines without this problem. `hdf5-java` should be the same package as `hdf5` just with the java bindings which are needed by `hdf5-view`. If you don't want or need `hdf5-view` then just switching to the `hdf5` package has no drawbacks. But if you do want `hdf5-view` you could try reinstalling `hdf5-java` and maybe the `python-h5py` package, but I'm not really sure why you should need to do this unless something else I don't know about changed in your system.