mutovis/control-software #20: Override voltage compliance limit

rbrenesh, 2019-06-07:
There is no way to change the voltage compliance limit to be above 2V, so modules cannot be tested properly.

greyltc, 2019-06-15:
this should fix it! 71af1054feaeb4825b1f25410b5f46da66bda83e
You can now use `--voltage-compliance-override 3.0` and I think that should do what you want. The issue was only with the initial Voc measurement, after that everything I do is in current source mode so there should be no more problems.

If I remember correctly, the relays I've used are good for up to 5A and up to +/- 60V. They'll probably be destroyed if they go above those limits (at an ambient temperate of 25 degC). There's some derating there though if the chips have some external heat load (like being in the sun or under a solar sim). So maybe talk to me if you'll get close to an amp.

I've never really tested these limits so let me know how it goes ;-) I expect the traces on the boards and/or the adapter board connector will burn up before you get to 5A though.

rbrenesh, 2019-08-01:
Tested and working! Sorry I didn't close it earlier

closed 2019-08-01 by rbrenesh