mutovis/control-software #21: User-definable voltage range

rbrenesh, 2019-08-01:
Some users in our lab have asked if it would be possible to define the voltage sweep range, since some want to bias beyond Voc and below 0V. Don't know if this would be possible to add to the software?

Also, but not as critical as the above, to allow for some pre-biasing times.

hollla, 2019-10-01:
[image]

having the ability to scan user-defined voltage is super important - currently, in some scans, it doesn't even pass both the Voc and Jsc, like the attached photo.

hollla, 2019-10-03:
Thank you @greyltc! ` [--scan-low-override SCAN_LOW_OVERRIDE]
                       [--scan-high-override SCAN_HIGH_OVERRIDE]` give me what I need.

This is what I run where I saw the scan didn't go both Jsc and Voc:

`mutovis-control --layout-index 12 --destination /home/onelab/data --pcb-address 192.168.1.11:23 --light-address ftdi://ftdi:232/1 --sm-address GPIB0::24::INSTR --rear true --four-wire on --pixel-address A1A2A3A4A5A6 --sweep on --snaith on --mppt 10 --mppt-params gradient_descent://0.001:0.001:10  -o DAK -r "buffalo thickness study" -p thickness 1.2m`

I did run the code on some cells with very minimal hysteresis and still observe the similar problem. 

Yes, it is possible it didn't do enough time of steady-state Voc measurement. With 10s of steady-state measurement, I mostly see the Voc is actually not fully stabilized (for example like attached photo).  
[image]

Anyway, thanks for the solution!

closed 2019-11-26 by rbrenesh