mutovis/control-software #23: Perform JV scans without stabilizing at Voc first

hollla, 2019-10-01:
Perform JV scans without stabilizing at Voc first would be helpful!

greyltc, 2019-10-01:
@hollla The steady state Voc measurement I do as the very first thing is pretty fundamental to the measurement routine. I use it to set sweep ranges and determine the device's polarity. Without knowing these things, several other parts of the code *might* end up breaking such as max power point tracking and the signage on the output data values.

Do you think it would give you what you need if I added a `--no-steady-state` option? If I did add that, you'd definitely need to specify
```
                       [--scan-low-override SCAN_LOW_OVERRIDE]
                       [--scan-high-override SCAN_HIGH_OVERRIDE]
```
Otherwise, I'd have no idea what ranges to sweep over!

`--no-steady-state` would mean that the measurement routine would not do the steady state Voc measurement at the start and it would also not do the steady state Isc measurement.

hollla, 2019-10-01:
I definitely agree with the importance of stabilizing Voc before taking the JV scans, but with all the weirdness of ionic perovskite properties, it is I think a consensus in the community that the JV scans should pass both Voc and Jsc. For example, Kamat et al (https://pubs.acs.org/doi/full/10.1021/acs.jpclett.5b00289) recommend a ~120%*Voc as starting bias voltage.

Or Henry's recent Nature paper:
https://www.nature.com/articles/s41586-019-1357-2
"The J–V curves were measured at a scan rate of 200 mV s−1 (delay time of 100 ms) from 1.2 V to −0.2 V and then back again (from −0.2 V to 1.2 V). A stabilization time of 2 s at forward bias of 1.2 V under illumination was done before scanning. "

I understand in the end, stabilized power output is all that matters, but having the ability to control scan range and rate is equally important to understand the devices we are making.

I will test `--no-steady-state` shortly.

Thank you a lot for the help!

greyltc, 2019-10-01:
`--no-steady-state` was just an idea! It doesn't exist. I was trying to ask you: if I added a feature like that, would it give you what you're looking for?

hollla, 2019-10-02:
Yes!