mutovis/control-software #25: Connection refused from board

swartr, 2021-12-16:
tried everything we can think of - the computer is only on a local network with the board:

mutovis-control --layout-index 12 --destination /home/onelab/data --pcb-address 192.168.1.11:23 --light-address ftdi://ftdi:232/1 --sm-address GPIB0::24::INSTR --rear false --four-wire on --pixel-address A1A2A3A4A5A6 --sweep on --snaith on --mppt 0 --mppt-params gradient_descent://0.001:0.001:10  -o rich -r "" -p Device "20210307" --scan-high-override 1.2 --scan-low-override -0.2
Software revision: 1.2.2
Using pyvisa-py pyvisa backend.
Sourcemeter found:
KEITHLEY INSTRUMENTS INC.,MODEL 2400,0637449,C34 Sep 21 2016 15:30:00/A02  /C/D
Traceback (most recent call last):
  File "/usr/bin/mutovis-control", line 11, in <module>
    load_entry_point('mutovis-control==1.2.2', 'console_scripts', 'mutovis-control')()
  File "/usr/lib/python3.7/site-packages/mutovis_control/__main__.py", line 12, in main
    c.run()
  File "/usr/lib/python3.7/site-packages/mutovis_control/cli.py", line 189, in run
    l.connect(dummy=args.dummy, visa_lib=args.visa_lib, visaAddress=args.sm_address, visaTerminator=args.sm_terminator, visaBaud=args.sm_baud, lightAddress=args.light_address, motionAddress=args.motion_address, pcbAddress=args.pcb_address, ignore_adapter_resistors=args.ignore_adapter_resistors)
  File "/usr/lib/python3.7/site-packages/mutovis_control/fabric.py", line 83, in connect
    self.pcb = pcb(address=pcbAddress, ignore_adapter_resistors=ignore_adapter_resistors)
  File "/usr/lib/python3.7/site-packages/mutovis_control/pcb.py", line 19, in __init__
    s.connect((ipAddress, int(port)))
ConnectionRefusedError: [Errno 111] Connection refused
Exception ignored in: <function pcb.__del__ at 0x7feef3d547b8>
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/mutovis_control/pcb.py", line 58, in __del__
    self.disconnect()
  File "/usr/lib/python3.7/site-packages/mutovis_control/pcb.py", line 74, in disconnect
    self.sf.close()
AttributeError: 'pcb' object has no attribute 'sf'
sys:1: ResourceWarning: unclosed <socket.socket fd=10, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('0.0.0.0', 40730)>

swartr, 2022-01-03:
I'm trying to see if re-flashing the Arduino mega changes anything.  Although I can't seem to find anything in the rest of this repository or the hardware repository as well.  Let me know where I can find it. 
This issue is preventing us from taking J-Vs using this system as its just refusing connection, so it would be nice to diagnose this problem quickly

rbrenesh, 2022-01-03:
@greyltc do you have any suggestions on how to troubleshoot this?

swartr, 2022-01-03:
onelab@onelab-pc ~]$ mutovis-control --layout-index 12 --destination /home/onelab/data --pcb-address 192.168.1.11:23 --light-address ftdi://ftdi:232/1 --sm-address GPIB0::24::INSTR --rear false --four-wire on --pixel-address A1A2A3A4A5A6 --sweep on --snaith on --mppt 0 --mppt-params gradient_descent://0.001:0.001:10  -o rich -r "" -p Device "20210307" --scan-high-override 1.2 --scan-low-override -0.2
Software revision: 1.2.2
Using pyvisa-py pyvisa backend.
Sourcemeter found:
KEITHLEY INSTRUMENTS INC.,MODEL 2400,0637449,C34 Sep 21 2016 15:30:00/A02  /C/D
Traceback (most recent call last):
  File "/usr/bin/mutovis-control", line 11, in <module>
    load_entry_point('mutovis-control==1.2.2', 'console_scripts', 'mutovis-control')()
  File "/usr/lib/python3.7/site-packages/mutovis_control/__main__.py", line 12, in main
    c.run()
  File "/usr/lib/python3.7/site-packages/mutovis_control/cli.py", line 189, in run
    l.connect(dummy=args.dummy, visa_lib=args.visa_lib, visaAddress=args.sm_address, visaTerminator=args.sm_terminator, visaBaud=args.sm_baud, lightAddress=args.light_address, motionAddress=args.motion_address, pcbAddress=args.pcb_address, ignore_adapter_resistors=args.ignore_adapter_resistors)
  File "/usr/lib/python3.7/site-packages/mutovis_control/fabric.py", line 83, in connect
    self.pcb = pcb(address=pcbAddress, ignore_adapter_resistors=ignore_adapter_resistors)
  File "/usr/lib/python3.7/site-packages/mutovis_control/pcb.py", line 19, in __init__
    s.connect((ipAddress, int(port)))
ConnectionRefusedError: [Errno 111] Connection refused
Exception ignored in: <function pcb.__del__ at 0x7f21494197b8>
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/mutovis_control/pcb.py", line 58, in __del__
    self.disconnect()
  File "/usr/lib/python3.7/site-packages/mutovis_control/pcb.py", line 74, in disconnect
    self.sf.close()
AttributeError: 'pcb' object has no attribute 'sf'
sys:1: ResourceWarning: unclosed <socket.socket fd=10, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('0.0.0.0', 56098)>

swartr, 2022-01-03:
the error I get after fully restarting the computer, board, switch, and keithley

swartr, 2022-01-03:
Used telnet 192.168.0.11 23 to check connections to the board.  Started randomly guessing and found the board was, for some reason, now on 192.168.0.10.  So now it all works

ip addr, ip maddr, ip neigh were no help in this matter - the port was listed as 192.168.0.11 on ip addr and didn't show up at all on ip maddr or neigh.

side note: We tried to install net-tools on manjaro but for some reason the package manager never seems to want to do it and it always fails.  Manjaro is a pretty tough OS to work with in this regard because it makes things difficult.

If the IP is going to be hard coded into the command line - it would be helpful to know a good way to scan the local network so that we don't waste time trying to find out where the board is.  Leaving open in case @greyltc has any suggestions

greyltc, 2022-01-17:
Hey, @swartr, very glad to hear that it's working okay now!

Please forgive me if some of my advise here isn't 100% accurate because I can't clearly remember exactly what your network setup is like, but here goes...
To prevent dynamic IP address allocation for any host in general, I think the best solution is to configure your DHCP server so that it will always issue the same address to the board. Your config change will be one that essentially tells your DHCP server to set up a static IP address allocation for the board by telling it something like "always assign IP address 192.168.0.11 (or whatever) to device with MAC address 34:54:DE:ED:33:44 (or whatever), and don't give that IP to any other device that might join the network" The details of this sort of config change is of course very DHCP server specific, so I think maybe I can't help beyond that.

That all said, [`nmap`](https://nmap.org/) is a command line linux tool that can scan subnets for devices.